robimarko/openwrt #50: Xiaomi AX3600 Brick After flash the latest update

omensight, 2022-08-08:
Hello, I have bricked my Router Xiaomi AX3600 after flashing the latest release, I have tried to unbrick my device with this guide without success https://openwrt.org/inbox/toh/xiaomi/xiaomi_ax3600#debricking, I would like to grab your help, thanks in advance.

robimarko, 2022-08-08:
You are gonna have to hook up UART

omensight, 2022-08-08:
> You are gonna have to hook up UART
Thanks, how could I do that? Do you have some guide I can use?

robimarko, 2022-08-08:
Its all documented in the wiki:
https://openwrt.org/inbox/toh/xiaomi/xiaomi_ax3600#serial

omensight, 2022-08-08:
Hi, I'm new in this topic, I have two questions please.
1. Can I use this TTL CP2102 (6Pin) ?
[image]
2. What do no ttl pin voltage 1.8V  mean?
Thanks in advance.

robimarko, 2022-08-08:
No, CP2102 is not 1.8V IO.

You must use a 1.8V IO adapter, otherwise you will fry the SoC

tm4dev, 2022-08-08:
> Hello, I have bricked my Router Xiaomi AX3600 after flashing the latest release, I have tried to unbrick my device with this guide without success https://openwrt.org/inbox/toh/xiaomi/xiaomi_ax3600#debricking, I would like to grab your help, thanks in advance.

Did you try miwifi repair tool? If you didnt modify partitions, I think it can help. It's easy way

omensight, 2022-08-08:
> Did you try miwifi repair tool? If you didnt modify partitions, I think it can help. It's easy way

Yes, I have tried it yesterday and the flash screen never popped up.

tm4dev, 2022-08-08:
> Yes, I have tried it yesterday and the flash screen never popped up.

yes I know, I have been in your situation. the problem is that tool did not change the network card configuration probably. take a look at the official instruction here https://www.xiaomi.cn/post/19134127

joba-1, 2022-08-08:
> 2. What do no ttl pin voltage 1.8V  mean?
>    Thanks in advance.

it means the sending/receiving lines expect this voltage and not more.

If you can‘t find one: I use this one:

€ 3,72 | FT232RL serial port module USB to TTL serial port small board 5V 3.3V 1.8V level
https://a.aliexpress.com/_vq2vab

omensight, 2022-08-08:
> > 2. What do no ttl pin voltage 1.8V  mean?
> >    Thanks in advance.
> 
> it means the sending/receiving lines expect this voltage and not more.
> 
> If you can‘t find one: I use this one:
> 
> € 3,72 | FT232RL serial port module USB to TTL serial port small board 5V 3.3V 1.8V level https://a.aliexpress.com/_vq2vab

So I have to get a 1.8v board? 
[image]
I think this one is not good in this case.

joba-1, 2022-08-08:
it needs to mention 1.8V or it will not have it - it is a not so common voltage.

robimarko, 2022-08-08:
FT232 ones are usually what you want as they are cheap and support 1.8V

omensight, 2022-08-08:
> FT232 ones are usually what you want as they are cheap and support 1.8V

I didn't found a 1.8v in my country and in the case I buy it from AliExpress it gonna arrive in my country in three months or more, is there another solution?

robimarko, 2022-08-09:
There is no way around getting a 1.8V UART adapter, other than frying your CPU

robimarko, 2022-08-09:
Great to hear that

closed 2022-08-09 by robimarko

w1ndy, 2022-08-19:
> There is no way around getting a 1.8V UART adapter, other than frying your CPU

Just adding my two cents here: Actually I've made successful attempts to repair my AX3600 with a 3.3V adapter... You'll have to start the router only with TXD and RXD connected and plug in GND immediately when the router LED starts to blink

[image]

robimarko, 2022-08-19:
That's really risky, pins are 1.8V rated and using 3.3V for any duration can easily fry it.