robimarko/openwrt #53: Implement ipq60xx support (shameless plug)

mrnuke, 2022-08-09:
I have a TP-Link EAP610-Outdoor that I'd like to support. The purpose of this shameless plug is to let people know I'm working on a ipq60xx device.

I've started implementing it here: https://github.com/mrnuke/openwrt/commits/eap610-outdoor

It's based heavily on the existing ipq807x work from this repo. I don't have a plan on this this might be merged, but since it relies on nss-packages for the time being, I would like it to go through this repo first. I'll happily takes comments, concerns or suggestions in this issue.

robimarko, 2022-08-09:
I have started poking around IPQ60xx a month ago, done some basic fixes that were quickly merged upstream.
CPU scaling is completely broken in the kernel, it is pending merge upstream.

Plenty of other stuff is missing as well, the issue is that I only have the CP01 dev board(A crappy Wallys one) and EAP101 as far as the HW goes.

mrnuke, 2022-08-09:
I'll leave the CPU scaling to the experts (that's you @robimarko).

I'm currentlly looking at:

-  Identify minimalist set of required patches
- Get ethernet working via nss-dp (scary: Realtek PHY has an upstream driver in 5.15)
- Get WiFi working on the assumption that working driver exists

robimarko, 2022-08-09:
ath11k should work, at least it worked for me couple of months ago

mrnuke, 2022-08-10:
How did you get the `q6v5_wcss: remoteproc@cd00000` node to load a driver? The `compatible = "qcom,ipq6018-wcss-pil";` property is in the upstream dtsi, yet I was not able to find a corresponding driver.

robimarko, 2022-08-10:
Well, like I said a ton of stuff is missing, WCSS is one of them.
You need the WCSS series from IPQ807x and then to add compatible in the driver for IPQ6018.
I am not sure if there are missing clocks as well

mrnuke, 2022-08-15:
I'm not quite getting the ethernet working with qca-nss-dp and qca-ssdk.. The driver loads, eth0 appears, but no packets go through -- ess switch, uniphy, and edma nodes are present in dtb. I noticed a couple of clock failures:

```
[    1.490360] ssdk_uniphy_port5_clock_source_set[1031]:INFO: ( nss_port5_rx_clk_src :: uniphy1_gcc_rx_clk )
[    1.490384] ssdk_uniphy_port5_clock_source_set[1037]:ERROR:set parent fail: -22!
[    1.499981] ssdk_uniphy_port5_clock_source_set[1031]:INFO: ( nss_port5_tx_clk_src :: uniphy1_gcc_tx_clk )
[    1.499999] ssdk_uniphy_port5_clock_source_set[1037]:ERROR:set parent fail: -22!
```

But, the clocks exist:

```
root@OpenWrt:/# ls /sys/kernel/debug/clk/uniphy1* -d
/sys/kernel/debug/clk/uniphy1_gcc_rx_clk
/sys/kernel/debug/clk/uniphy1_gcc_tx_clk
root@OpenWrt:/# ls /sys/kernel/debug/clk/nss_port5* -d
/sys/kernel/debug/clk/nss_port5_rx_clk_src
/sys/kernel/debug/clk/nss_port5_rx_div_clk_src
/sys/kernel/debug/clk/nss_port5_tx_clk_src
/sys/kernel/debug/clk/nss_port5_tx_div_clk_src
```

Is this something you've seen before?

robimarko, 2022-08-15:
As I said before, there are clocks missing or broken for sure.
Resets for networking are missing for sure, that I know of.

solidus1983, 2022-08-18:
I have IPQ6018 GL-Flint i could help test with on this with UART connection.

mrnuke, 2022-08-19:
@solidus1983, I have some peripherals working on my eap610-outdoor branch, to which I force-push regularly.

Wifi and ethernet are not working. Board reboots when enabling WiFi. Ethernet via nss-dp-drv complains about some clocks and doesn't send any interface traffic -- or there's a kernel panic related to edma. Trying to follow @robimarko's advice about locating the missing bits.

solidus1983, 2022-08-19:
@mrnuke Thanks for the reply, i take it this is not using the QSDK kernel 5.4? Also could clocks be in the dts file ? if so maybe looking at another IPQ6018 SOC's dts which has working clocks might help no ?

Else the offer is still open for testing. I have yet to start coding and currently learning. I take it this is C# being used ?

What i do know is that GL-Inet has been using the IPQ807x drivers for IPQ6018 so it might be a good start there possibly.

robimarko, 2022-08-19:
DTS is just a way to describe the HW, it doesn't actually implement anything.
There is a lot of clock related stuff missing, PLLs are using outdated config etc

solidus1983, 2022-08-19:
@robimarko I am new to all this to be honest, so thinking out side the box so to speak.

mrnuke, 2022-08-28:
@robimarko, I'm trying to understand what they did in the QCA 5.4 downstream kernel. Cherry-picking a few clock commits fixes a lot of issues. I'm not understanding why they would change the nss_port clk parents from `.parent_data` to `.parent_names`. 

The claim is it fixes a crash, but I see it fixes the `Clk set parent fail!` complaint from qca-ssdk.

https://git.codelinaro.org/clo/qsdk/oss/kernel/linux-ipq-5.4/-/commit/d7022cd843188144ec6bf7e996552d2104da4a62

robimarko, 2022-08-28:
Ughh, they messed up like they always do, and instead of fixing that they just hack it.
I assume they messed up the parenting as they only set `fw_name` which is locally available, but SSDK is trying to set parenting manually without them being available as you need to pass them by DT node.

So, what probably needs to be done instead of hacking it together and itself is a workaround for a lack of proper switch driver is to also set the `.name` property so it will still fallback to global clock names

mrnuke, 2022-08-28:
You're right, just doing the following fixes the parenting issue
``` patch
       { .fw_name = "uniphy0_gcc_rx_clk", .name = "uniphy0_gcc_rx_clk" },
       { .fw_name = "uniphy0_gcc_tx_clk", .name = "uniphy0_gcc_tx_clk" },
       { .fw_name = "uniphy1_gcc_rx_clk", .name = "uniphy1_gcc_rx_clk" },
       { .fw_name = "uniphy1_gcc_tx_clk", .name = "uniphy1_gcc_tx_clk" },
```

I noticed in the ipq807x branch, there are some ipq6018 fixes (which I promptly picked up). How can we keep my ports and your work in sync? I'm thinking if maybe there's a branch I can send you PR's to.

robimarko, 2022-08-29:
Those are for the PLL so that clock scaling actually works instead of the broken sh*t QCA upstreamed like it never even worked from the start.

I dont have an OpenWrt tree at all as all I did on IPQ60xx was targeted at linux-next, my plan was to get working on it again after finishing IPQ807x which as you can tell is dragging on forever due to ath11k issues

mrnuke, 2022-08-30:
It can be a kernel branch. I mostly do `git format-patch -o ../openwrt/target/linux/ipq60xx/patches-5.15/` anyway. I want to make sure I have the things you consider important, and make sure you don't forward-port changes that I already cherry-picked.

mrnuke, 2022-09-04:
I'm a bit stuck. On top of all the patches for ip807x-5.15 branch, I took changes form the QCA 5.4 kernel, related to clocks and wcss:

- [714e6b9ab711 "clk: ipq6018: Add missing clocks "](https://git.codelinaro.org/clo/qsdk/oss/kernel/linux-ipq-5.4/-/commit/714e6b9ab711846cd09b94f83436e23932dd2814)
- [5b6c148bfb6c "ipq6018: rproc: Add non secure Q6 bringup sequence "](https://git.codelinaro.org/clo/qsdk/oss/kernel/linux-ipq-5.4/-/commit/5b6c148bfb6c25449c59a0a87896a90f883307fd)
- other misc patches relating to these two peripherals

Problem 1 is that, while nss-dp loads up and does not crash, the "TX bytes" of `eth0` always shows up as 0.0 B. I can check with `tcpdump` that packets are received successfully. I also see packets being sent, but I can't see those packets on the other end of the wire.

Problem 2 is that the remoteproc crashes due to a watchdog. I suspected that might be clock related and set `.need_mem_protection = false,` to bring up up more clocks. ath11k then fails with `Coldboot Calibration failed`. IN either case, the system then unexpectedly reboots and hangs.

solidus1983, 2022-09-11:
So after looking at the current work that @mrnuke has done whats the best way to port the GL-AX1800 over.
I have tried already with simply dumping in the QCA 5.4 Kernel dts and dtsi files over and that didn't compile well.

I am rather new to all this however i have UART on the board and willing to learn.

mrnuke, 2022-09-11:
What I've done so far was to diff against QCA 5.4 kernel, and see what patches I need to cherry-pick. You can use most of the nodes from the `eap610` dts. The `ess-switch` and `dp@5` nodes might need some adjustment for your device.
You might need to extract the factory firmware for that info, as well as the factory dtb, and board-specific wifi firmware. Once you plug in those pieces, we have to figure out why Ethernet isn't transmitting, and why wifi crashes.

solidus1983, 2022-09-11:
@mrnuke Well currently i build against the GL-iNet infrabuilder repo which has the dts/dtsi files and patches however doesn't seem to play nice with OpenWRT might be due to it being QCA 5.4 Kernel.

As for the Wireless crashes, when does it crash before or after it's been brought up?
As for Ethernet no transmitting surely, that would be a driver issue.

mrnuke, 2022-09-11:
The first step to get this running on OpenWRT is to get it running with an upstream kernel -- currently 5.15). You can start on top of my eap610-outdoor branch -- I force-push to it regularly.

### Extracting vendor firmware

Good news! The GL-iNet images are simple QSDK upgrade images. You can analyze them easily:

```bash
$ dumpimage -V
dumpimage version 2022.04
$ dumpimage -l qsdk-ax1800-3.214-0509.img 
Image contains unit addresses @, this will break signing
FIT description: Flashing nand 800 20000
Created:         Sun May  8 23:23:36 2022
 Image 0 (script)
```

#### Filesystem
    
```
pip3 install ubi-reader --user
```
```
$ dumpimage qsdk-ax1800-3.214-0509.img -T flat_dt -p 4 -o qsdk-ubi
$ ubireader_extract_images qsdk-ubi 
$ ls -lh ubifs-root/qsdk-ubi/
-rw-r--r--. 1 mrnuke mrnuke 4.7M Sep 11 13:30 img-874724779_vol-kernel.ubifs
-rw-r--r--. 1 mrnuke mrnuke    0 Sep 11 13:30 img-874724779_vol-rootfs_data.ubifs
-rw-r--r--. 1 mrnuke mrnuke  31M Sep 11 13:30 img-874724779_vol-ubi_rootfs.ubifs
-rw-r--r--. 1 mrnuke mrnuke 2.0M Sep 11 13:30 img-874724779_vol-wifi_fw.ubifs
```

##### Kernel and devictree

```
$ dumpimage ubifs-root/qsdk-ubi/img-874724779_vol-kernel.ubifs -T flat_dt -p 7 -o extracted.dtb 
$ dtc -i dtb extracted.dtb 
```

The correct dtb to extract depends on the configuration -- usually printed on the serial console on boot,

##### Wi-Fi Firmware

```
$ unsquashfs ubifs-root/qsdk-ubi/img-874724779_vol-wifi_fw.ubifs
$ ls squashfs-root/
bdwlan.b00  bdwlan.b20  bdwlan.bin                m3_fw.flist  q6_fw.b04    qdss_trace_config.bin
bdwlan.b01  bdwlan.b21  firmware_rdp_feature.ini  m3_fw.mdt    q6_fw.b05
bdwlan.b02  bdwlan.b22  fw_version.txt            q6_fw.b00    q6_fw.b07
bdwlan.b10  bdwlan.b23  m3_fw.b00                 q6_fw.b01    q6_fw.b08
bdwlan.b11  bdwlan.b30  m3_fw.b01                 q6_fw.b02    q6_fw.flist
bdwlan.b12  bdwlan.b31  m3_fw.b02                 q6_fw.b03    q6_fw.mdt
```

solidus1983, 2022-09-11:
Will dump the v4 firmware as thats Kernel 5.4 and not Kernel 4.4.46. Will print the results on here.

mrnuke, 2022-09-11:
> Using 'config@cp03-c1' configuration

Okay, so `dumpimage` should list a configuration named `config@cp03-c1`, which has the corrrect dtb name. There should also an image node with the dtb name found.

```
$  dumpimage ubifs-root/qsdk-ubi/img-874724779_vol-kernel.ubifs -T flat_dt -p 7 -o extracted.dtb
...
 Image 7 (fdt@cp03-c1)
  Description:  ARM OpenWrt qcom-ipq60xx-cpxx device tree blob
...
 Configuration 6 (config@cp03-c1)
  Description:  OpenWrt
  Kernel:       kernel@1
  FDT:          fdt@cp03-c1
```
In this case it happens to be image 7 that has the correct DTB. The previous instructions give image 7 as the example, so should work.

My next suggestion is to get the **ess-switch** and **dp1** -> **dp5** nodes in the OpenWRT devicetree to look like the nodes in the extracted dtb. Note that **dp** nodes in openwrt use `phy-handle` instead of the `qcom,reverse-rhodesian` properties in the QSDK.

Does the vendor u-boot allow you to enter a console and boot from tftp? If so, would be interesting to feed it an initramfs image, and see if networking works on your device, or if it fails on TX.

solidus1983, 2022-09-11:
Yes i can break uboot with ```gl``` to boot tftp, host httpd the works.

mrnuke, 2022-09-11:
```
setenv ipaddr 192.168.7.69
setenv serverip 192.168.7.2
tftpboot openwrt-ipq.img
bootm
```
Replace `serverip` and `ipaddr` with your setuo.

solidus1983, 2022-09-11:
> ```
> setenv ipaddr 192.168.7.69
> setenv serverip 192.168.7.2
> tftpboot openwrt-ipq.img
> bootm
> ```
> 
> Replace `serverip` and `ipaddr` with your setuo.

Ok, where is the openwrt-ipq.img ?

mrnuke, 2022-09-11:
That's the openwrt image you put on your tftp server:

``` 
$ tftp <serverip> -c put bin/targets/ipq60xx/generic/openwrt-ipq60xx-generic-eap610-outdoor-initramfs-fit-uImage.itbopenwrt-ipq.img
```

solidus1983, 2022-09-11:
> 

Do you have a copy of that image ? Saves booting the compiler up.

mrnuke, 2022-09-11:
My eap610 image would be pretty useless without the devicetree adjustments described earlier

solidus1983, 2022-09-11:
> My eap610 image would be pretty useless without the devicetree adjustments described earlier

Ah right will get on that later if i have free time. Thanks for you help though.

kirdesde, 2022-09-18:
> I'm a bit stuck. On top of all the patches for ip807x-5.15 branch, I took changes form the QCA 5.4 kernel, related to clocks and wcss:
>
>     * [714e6b9ab711 "clk: ipq6018: Add missing clocks [...]

Hi. thanks for working on the ipq6018 target.
I've got ath11k working (on a netgear WAX610)

1. coldboot calibration needs to disabled (like for the IPQ8074)
`--- a/drivers/net/wireless/ath/ath11k/core.c
+++ b/drivers/net/wireless/ath/ath11k/core.c
@@ -158,7 +158,7 @@ static const struct ath11k_hw_params ath
 		.supports_shadow_regs = false,
 		.idle_ps = false,
 		.supports_sta_ps = false,
-		.cold_boot_calib = true,
+		.cold_boot_calib = false,
 		.fw_mem_mode = 0,
 		.num_vdevs = 16 + 1,
 		.num_peers = 512,
 		`
2. I'm using the following firmware from the quic upstream repo: WLAN.HK.2.6.0.1-00861
The 2.5.01 firmware led to a ath11k crash related to regulatory update.

solidus1983, 2022-09-19:
> > ```
> > setenv ipaddr 192.168.7.69
> > setenv serverip 192.168.7.2
> > tftpboot openwrt-ipq.img
> > bootm
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Replace `serverip` and `ipaddr` with your setuo.
> 
> Ok, where is the openwrt-ipq.img ?

```Wrong Image Format for bootm command
ERROR: can't get kernel image!```

It simply give me the error above.

mrnuke, 2022-09-19:
>     1. coldboot calibration needs to disabled (like for the IPQ8074)

**Thanks you!** That's what was causing most of the problems!

[image]

>     2. I'm using the following firmware from the quic upstream repo: WLAN.HK.2.6.0.1-00861
>        The 2.5.01 firmware led to a ath11k crash related to regulatory update.

Once I got fix #1 in, I was able to access the device via wifi, despite an inordinate amount of nagging from ath11k. I got the new board-2.bin from [quic repo here](https://git.codelinaro.org/clo/qsdk/oss/ath11k-bdf/-/blob/NHSS.QSDK.12.1.5.r3/IPQ6018/hw1.0/WLAN.HK.2.6.0.1/WLAN.HK.2.6.0.1-00861-QCAHKSWPL_SILICONZ-1)

> BTW And the DP is working fine for me.

Let me guess. WAX610 uses a WCA phy that's supported in qca-ssdk, not some realtek variant, right?

kirdesde, 2022-09-20:
> > ```
> > 1. coldboot calibration needs to disabled (like for the IPQ8074)
> > ```
> 
> **Thanks you!** That's what was causing most of the problems!
> 
> [image]
> 
> > ```
> > 2. I'm using the following firmware from the quic upstream repo: WLAN.HK.2.6.0.1-00861
> >    The 2.5.01 firmware led to a ath11k crash related to regulatory update.
> > ```
> 
> Once I got fix 1 in, I was able to access the device via wifi, despite an inordinate amount of nagging from ath11k. I got the new board-2.bin from [quic repo here](https://git.codelinaro.org/clo/qsdk/oss/ath11k-bdf/-/blob/NHSS.QSDK.12.1.5.r3/IPQ6018/hw1.0/WLAN.HK.2.6.0.1/WLAN.HK.2.6.0.1-00861-QCAHKSWPL_SILICONZ-1)
> 
> > BTW And the DP is working fine for me.
> 
> Let me guess. WAX610 uses a QCA phy that's supported in qca-ssdk, not some realtek variant, right?

Yeah, it's a QCA8081

mrnuke, 2022-09-20:
<details open>
<summary>Serial log of EAP610-Outdoor on vendor Firmware</summary>
<pre>
IPQ6018# setenv bootargs
IPQ6018# bootipq
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=0", size 33 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 269, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 2, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1172702875
ubi0: available PEBs: 122, total reserved PEBs: 147, PEBs reserved for bad PEB handling: 20
checking volume(ubi_rootfs) passed
checking volume(kernel) passed
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (3691040)
fdt_getprop find dtb_config_name config@cp03-c1
bootm 0x44000000#config@cp03-c1
## Loading kernel from FIT Image at 44000000 ...
   Using 'config@cp03-c1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  ARM64 OpenWrt Linux-4.4.60
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x440000e8
     Data Size:    3616720 Bytes = 3.4 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x41080000
     Entry Point:  0x41080000
     Hash algo:    crc32
     Hash value:   8dc1f352
     Hash algo:    sha1
     Hash value:   858dae32e65cb82e6e81305c9a257d008fd4b2ef
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 44000000 ...
   Using 'config@cp03-c1' configuration
   Trying 'fdt@cp03-c1' fdt subimage
     Description:  ARM64 OpenWrt qcom-ipq6018-cp03-c1 device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x44373200
     Data Size:    71440 Bytes = 69.8 KiB
     Architecture: AArch64
     Hash algo:    crc32
     Hash value:   0ab25e08
     Hash algo:    sha1
     Hash value:   2c7b142b854aa98a1b5ea46674528bebe3d4bbca
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x44373200
   Uncompressing Kernel Image ... OK
   Loading Device Tree to 484eb000, end 484ff70f ... OK
Could not find PCI in device tree
Using machid 0x8030200 from environment
Starting kernel ...
Jumping to AARCH64 kernel via monitor
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.60 (jenkins@sohoiapbuild) (gcc version 5.2.0 (OpenWrt GCC 5.2.0 ff0b127+r49254) ) #1 SMP PREEMPT Wed Jan 26 10:54:35 CST 2022
[    0.000000] Boot CPU: AArch64 Processor [51af8014]
[    0.000000] Ignoring memory range 0x40000000 - 0x41000000
[    0.000000] Machine: Qualcomm Technologies, Inc. IPQ6018/AP-CP03-C1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] psci: probing for conduit method from DT.
[... 837 lines cut ...]
iptables: Bad rule (does a matching rule exist in that chain?).
now ok to start tddp---------------------
uclite init ok, now startup eap-cs ---------------------
httpMudCreate: MUD 0x4a7e58 was created
[NM_Debug](nm_region_getRegionName) 00192: Flash region info, code: 841, name: US.
[NM_Debug](nm_region_getRegionName) 00192: Flash region info, code: 841, name: US.
[utilities_debug: httpSystemFirmwareInit:271]register rpm
[utilities_debug: httpSystemFirmwareInit:271]register rpm
httpServerCreate------------------
httpServerCreate------------------
httpServerCreate: try to add port 22080
httpServerCreate: try to add port 80
[Debug ledListenEventHandler:148] Accept a new client.
[Debug ledClientEventHandler:110] GPIOD received led rule: LED[   20.704507] [Debug led_proc_write:651] Write led_green.
[   20.708813] [Debug led_common_write_proc:490] Execute LED action:    { 7   0   0   0   0 }
_ISOLATED_FINISH.
[   28.691104] tp dfs events being sent to PID:284
route: SIOCDELRT: No such process
[UNIX_SOCK][unix_sock_connSrv:301]connect to file(/var/run/srv/22) failed after retry(0), errno(2):No such file or directory
Into util_dbg_setMod, pModName(all), enable(1)
route: SIOCDELRT: No such process
[UNIX_SOCK][unix_sock_connSrv:301]connect to file(/var/run/srv/22) failed after retry(0), errno(2):No such file or directory

</pre>
</details>

kirdesde, 2022-09-22:
For me it looks like they added the rtl phy init into the ssdk:

```
[  5.946201] HPPE initializing...
[    5.946904] rtl8211_phy_init[536]:ERROR:Init RTL8211, dev_id: 0, port_bmp: 20
[    5.953874] rtl8211_phy_api_ops_init[516]:INFO:qca probe rtl8211 phy driver succeeded!
```
Any chance for getting the gpl source?

Maybe @robimarko  has an idea how to add the phy init to the ssdk.

robimarko, 2022-09-22:
@kirdesde That is from the stock FW?

If so, its definitively not in the "upstream" SSDK.
BTW: https://git.codelinaro.org/clo/qsdk/oss/lklm/qca-ssdk/-/commit/c1cb29074905dae1f9a6d1f61ed601727b52c5c2

As far as I can tell, it should parse stuff from the kernel instead of internally just for cases where PHY driver is in kernel

kirdesde, 2022-09-22:
> @kirdesde That is from the stock FW?
Yes, it's from the provide stock kernel log.
> 
> If so, its definitively not in the "upstream" SSDK. BTW: https://git.codelinaro.org/clo/qsdk/oss/lklm/qca-ssdk/-/commit/c1cb29074905dae1f9a6d1f61ed601727b52c5c2
> 
> TP-Link has GPL for that EAP610: https://static.tp-link.com/upload/gpl-code/2022/202204/20220411/ipq60xx_eap610_outdoor_v1.tar.gz

Yeah, already saw that, but it doesn't contain any qca source related....I can only find prebuild kernel modules (inkl. qca*)
> 
> As far as I can tell, it should parse stuff from the kernel instead of internally just for cases where PHY driver is in kernel

kirdesde, 2022-09-22:
Looks like they only provide the bare minimum gpl sources, but anything else is pre-built (it contains an entire rootfs)

robimarko, 2022-09-22:
Then you should contact them, they must provide all of these components in source as they are GPLv2

mrnuke, 2022-09-23:
> BTW: https://git.codelinaro.org/clo/qsdk/oss/lklm/qca-ssdk/-/commit/c1cb29074905dae1f9a6d1f61ed601727b52c5c2

Excellent! That's just the patch I needed! I'll need to rebase your qca-ssdk patches on top of the newer release, but that fixes it!

robimarko, 2022-09-23:
Excellent, it also allows to move from the built-in drivers for other PHY-s as well in theory

kirdesde, 2022-09-23:
> Excellent, it also allows to move from the built-in drivers for other PHY-s as well in theory

You mean we can drop nss-dp and use the kernel drivers for QCA8075, QCA8081 and so on?

Or is the DP driver still necessary?

robimarko, 2022-09-23:
NSS-DP is the ethernet controller driver, you must use it, but it in theory allows not using the PHY drivers from SSDK.
For QCA8081 that isn't hard, but QCA8075 is a real mess

mrnuke, 2022-09-23:
I got the SSDK updated in PR #68

solidus1983, 2022-10-01:
@mrnuke 
```

TFTP from server 192.168.132.2; our IP address is 192.168.132.1
Filename 'ipq6018.img'.
Load address: 0x44000000
Loading: *
Got TFTP_OACK: TFTP remote port: changes from 69 to 59086
#################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ###########
         4.2 MiB/s
done
Bytes transferred = 9693213 (93e81d hex)
ipq6018_eth_halt: done
IPQ6018#
```

Then bootm throws

`ipq6018_eth_halt: done
IPQ6018# bootm
Wrong Image Format for bootm command
ERROR: can't get kernel image!
IPQ6018#
`

mrnuke, 2022-10-01:
@solidus1983, that might be a bit harder to track down. I only have a couple of suggestions:

- double-check that ipq6018.img is indeed the initramfs image
- Make sure you don't have a gateway at 192.168.132.1, or change the ip (`setenv ipaddr 192.168.132.xxx`)

solidus1983, 2022-10-01:
@mrnuke 

- Original file openwrt-ipq60xx-generic-gl-ax1800-initramfs-fit-uImage.itb
- TFTP Server is setup to 192.168.132.2 with direct connection from Flint to Laptop with no gateway.
- Routers ip is simply 192.168.132.1

I got as far as it is compiling just hitting the booting wall

kirdesde, 2022-10-01:
I'd also bet you are not booting a initramfs image. You are tftp-ing a ipq6018.img...

Please provide your image definition (generic.mk/Makefile)

solidus1983, 2022-10-01:
@kirdesde 

https://raw.githubusercontent.com/solidus1983/openwrt/eap610-outdoor/target/linux/ipq60xx/image/generic.mk
https://raw.githubusercontent.com/solidus1983/openwrt/eap610-outdoor/target/linux/ipq60xx/image/Makefile

Hope this helps, also i am new to the whole porting game.

kirdesde, 2022-10-01:
And ipq6018.img is just the renamed openwrt-ipq60xx-generic-gl-ax1800-initramfs-fit-uImage.itb?
Please do a binwalk of your ipq6018.img and post the output.

solidus1983, 2022-10-01:
@kirdesde yep, Will quickly boot the vm back up now for the binwalk what commands am i using ?

kirdesde, 2022-10-01:
just install the binwalk package and do

```
binwalk /path/to/ipq6018.img
```

solidus1983, 2022-10-01:
@kirdesde 

Binwalk output

`0             0x0             gzip compressed data, maximum compression, from Unix, last modified: 1970-01-01 00:00:00 (null date)`

kirdesde, 2022-10-01:
Your initramfs image is missing the FDT. It should look like this:

```DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Flattened device tree, size: 12044740 bytes, version: 17
232           0xE8            gzip compressed data, maximum compression, from Unix, last modified: 1970-01-01 00:00:00 (null date)
10978265      0xA783D9        AES S-Box
11739480      0xB32158        GIF image data, version "89a", 32 x
12019368      0xB766A8        Flattened device tree, size: 24016 bytes, version: 17
```

Do you even have a dtb in the build_dir/kernel_dir?

Please do a 

```
make target/install V=s
```
and look in the output for the initramfs generation, I guess there is a at least a warning regarding the missing dtb

solidus1983, 2022-10-01:
@kirdesde here is a log of building with the command you asked.

[error.txt](https://github.com/robimarko/openwrt/files/9690667/error.txt)

kirdesde, 2022-10-01:
```
Error: /home/openwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/linux-5.15.69/arch/arm64/boot/dts/qcom/ipq6018-gl-ax1800.dtsi:175.30-31 syntax error
FATAL ERROR: Unable to parse input tree
make[4]: *** [Makefile:19: /home/openwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/image-ipq6018-gl-ax1800.dtb] Error 1
```

solidus1983, 2022-10-01:
So looks like i can't use the TIP dts/dtsi files 😞

kirdesde, 2022-10-01:
> So looks like i can't use the TIP dts/dtsi files disappointed

At least not just copy & paste. It's better to use the eap610 dts as an example and then add / update the ax1800 specific stuff.

[68 more replies]

closed 2023-01-16 by robimarko