robimarko/openwrt #55: [TODO] Tracking stuff required for PR

robimarko, 2022-08-20:
This is a summary issue to track blockers from making a PR for OpenWrt inclusion.

**Remaining TODO**:
- [ ] Redo the kernel tree using upstreamed patches
- [ ] Move SSDK and NSS-DP out of feed into the main tree
- [ ] Single FW on Xiaomi devices
- [ ] SMP affinity script for the target
- [ ] ath11k needs to be fixed to stop breaking sysupgrade
- [ ] ath11k board-ids need to be dropped and BDF-s repackaged without them (Result of upstream conversation)
- [ ] Probably ath11k inclusion cleanup as we depend on the target before it even exists and so on

Ansuel, 2022-08-20:
> * ath11k board-ids need to be dropped and BDF-s repackaged without them (Result of upstream conversation)

i didn't understand this.

Also we are missing `enlarge rootfs size to the REAL end of the nand for Xiaomi AX3600`

> * Move SSDK and NSS-DP out of feed into the main tree

Wonder if we can work this without the need of a ipq807x pr? And start the discussion early?

robimarko, 2022-08-20:
> > * ath11k board-ids need to be dropped and BDF-s repackaged without them (Result of upstream conversation)
> 
> i didn't understand this.

The board-id that we manually set in DTS as nobody bothered to set it, I tried upstreaming that but we concluded that its
not required at all as we are setting the variant anyway.
So BDF-s need to be repackaged without it.

> 
> Also we are missing `enlarge rootfs size to the REAL end of the nand for Xiaomi AX3600`

It's part of the `Single FW on Xiaomi devices`
> 
> > * Move SSDK and NSS-DP out of feed into the main tree
> 
> Wonder if we can work this without the need of a ipq807x pr? And start the discussion early?

I dont see a reason for it, I plan to just make it a part of the PR as they are not optional.

Ansuel, 2022-08-20:
> I dont see a reason for it, I plan to just make it a part of the PR as they are not optional.\nSSDK and the \"GMAC\" kmods were already added before for IPQ806x

Idea is to not have a gigantic pr and try to review commit in smaller branch. Seems a good idea to have a separate pr for ssdk and dp. (Also to not have 50 comments on 4 different topic in one pr)

robimarko, 2022-08-20:
Sometimes that is not possible, we are talking about a whole new target here.
How can SSDK and NSS-DP depend on ipq807x if it doesn't exist?

kirdesde, 2022-08-20:
>  ath11k board-ids need to be dropped and BDF-s repackaged without them (Result of upstream conversation)

I can take care of that.
I understand, we remove the board-id from the dts and the bdf and keeping the variant in both, right?

robimarko, 2022-08-20:
Yes, we remove just the board-id and use the one that was burned from factory (Usually the default 0xff)

kirdesde, 2022-08-20:
> 

But we stick to one BDF per device? Or should we compile one BDF for all supported devices?

robimarko, 2022-08-20:
Yes, as ipq-wifi cannot handle anything else.
We just need to get rid of the board-id overrides and use whatever was baked in factory, some devices even now use `0xff` as vendor did not even override it themselves

kirdesde, 2022-08-20:
> Yes, as ipq-wifi cannot handle anything else. We just need to get rid of the board-id overrides and use whatever was baked in factory, some devices even now use `0xff` as vendor did not even override it themselves

Seems like ath11k needs the qmi-board-id in the BDF (I've remove the qmi-board-id from the BDF):

`[    7.702875] ath11k c000000.wifi: ipq8074 hw2.0
[    8.892778] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[    8.900237] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[    8.900283] ath11k c000000.wifi: fw_version 0x270204a5 fw_build_timestamp 2022-08-04 13:05 fw_build_id WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1
[    9.015110] ath11k c000000.wifi: failed to fetch board data for bus=ahb,qmi-chip-id=0,qmi-board-id=255,variant=Dynalink-DL-WRX36 from ath11k/IPQ8074/hw2.0/board-2.bin
[    9.015168] ath11k c000000.wifi: failed to fetch board data for bus=ahb,qmi-chip-id=0,qmi-board-id=255 from ath11k/IPQ8074/hw2.0/board-2.bin
[    9.029031] ath11k c000000.wifi: failed to fetch board.bin from IPQ8074/hw2.0
[    9.041666] ath11k c000000.wifi: qmi failed to fetch board file: -12
[    9.048665] ath11k c000000.wifi: failed to load board data file: -12`

board-2.json:

`
    {
        "names": [
            "bus=ahb,qmi-chip-id=0,variant=Dynalink-DL-WRX36"
        ],
        "data": "bus=ahb,qmi-chip-id=0,variant=Dynalink-DL-WRX36.bin"
    }
]`

So I think we can only remove the board-id from the dts, but leave the BDF's untouched.

robimarko, 2022-08-20:
I think we misunderstood each other.
I am not talking about completely removing it, but rather removing the DTS override and changing it back to whatever was burned in factory to the BDF.

So, whatever ath11k reports as `board_id` during boot with the DTS override removed as its a required part of board parsing

kirdesde, 2022-08-20:
> I think we misunderstood each other. I am not talking about completely removing it, but rather removing the DTS override and changing it back to whatever was burned in factory to the BDF.
> 
> So, whatever ath11k reports as `board_id` during boot with the DTS override removed as its a required part of board parsing. I think all of our boards pretty much will report `0xff` as I have seen only Qualcomm actually burn it

All right, but then I would need a ath11k boot log from all supported devices to make sure, what board_id they are requesting....

Or we rely on your assumption and set the default 255 in all BDF's...

robimarko, 2022-08-20:
It's actually simpler, just check the DTS-es, if it's getting overridden factory set `0xff` then, Dynalink also uses that despite me forgetting a comment, 301w uses it as well and its documented in the DTS.

So, as far as I can see, you can just set 255 aka `0xff` in all BDF-s and it will be good

kirdesde, 2022-08-20:
> It's actually simpler, just check the DTS-es, if it's getting overridden factory set `0xff` then, Dynalink also uses that despite me forgetting a comment, 301w uses it as well and its documented in the DTS.
> 
> So, as far as I can see, you can just set 255 aka `0xff` in all BDF-s and it will be good

All right, will do.

robimarko, 2022-08-20:
Please make it a separate commit for each device, so I can just squash it

kirdesde, 2022-08-20:
> Please make it a separate commit for each device, so I can just squash it

What do you think about that commit description?

`remove dts property "board_id" and
set qmi-board-id in the	BDF to the default 255/0xff
as per discussion:
https://lore.kernel.org/all/87o7yeo104.fsf@kernel.org/

robimarko, 2022-08-20:
Since I am just gonna squash them to the commit adding each of the devices, the description doesn't really matter.

robimarko, 2022-08-20:
@Ansuel We really need to figure out something to make sysupgrade reliable as ath11k is constantly breaking it since it doesn't let hostapd terminate

Ansuel, 2022-08-20:
I honestly think we need to add a workaround and ignore waiting for hostapd
to terminate or increase the wait time...

Root of the problem is ath11k being slow to destroy each peer

robimarko, 2022-08-20:
You cannot ignore it as it must terminate in order to switch the root to tmpfs and actually do the upgrade.

kirdesde, 2022-08-20:
What about adding an additional function in the platform.sh like "service stop wpad" + a check for the hostapd process?
According to some forum members "service wpad stop" was successfull, at least as an manual step before sysupgrade.

edrikk, 2022-08-20:
Yes a simple /etc/initd./wpad stop has resulted in a 100% success rate for me over the last several months (my own builds keeping up with master, so not just robi’s builds.

kirdesde, 2022-08-20:
This should do as a smp_affinity script (based on QSDK smp_affinity script and dchards script)

`#!/bin/sh /etc/rc.common

START=99

enable_affinity_ipq807x(){
#assign 4 rx interrupts to each cores
        irq_affinity_num=$(grep -E -m1 'reo2host-destination-ring1' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 1 > /proc/irq/$irq_affinity_num/smp_affinity
        irq_affinity_num=$(grep -E -m1 'reo2host-destination-ring2' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 2 > /proc/irq/$irq_affinity_num/smp_affinity
        irq_affinity_num=$(grep -E -m1 'reo2host-destination-ring3' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 4 > /proc/irq/$irq_affinity_num/smp_affinity
        irq_affinity_num=$(grep -E -m1 'reo2host-destination-ring4' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 8 > /proc/irq/$irq_affinity_num/smp_affinity

#assign 3 tcl completions to last 3 CPUs
        irq_affinity_num=$(grep -E -m1 'wbm2host-tx-completions-ring1' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 2 > /proc/irq/$irq_affinity_num/smp_affinity
        irq_affinity_num=$(grep -E -m1 'wbm2host-tx-completions-ring2' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 4 > /proc/irq/$irq_affinity_num/smp_affinity
        irq_affinity_num=$(grep -E -m1 'wbm2host-tx-completions-ring3' /proc/interrupts | cut -d ':' -f 1 | tail -n1 | tr -d ' ')
        [ -n "$irq_affinity_num" ] && echo 8 > /proc/irq/$irq_affinity_num/smp_affinity

}

boot() {
	case $(board_name) in
    dynalink,dl-wrx36|\
	edgecore,eap102|\
	edimax,cax1800|\
	qnap,301w|\
	redmi,ax6|\
	xiaomi,ax3600|\
	xiaomi,ax9000)
    enable_affinity_ipq807x
    ;;
	esac
}`

robimarko, 2022-08-21:
Ok, so it looks like the sysupgrade limit to for TERM signal is 10 tries, but for KILL its only 1.

Ansuel, 2022-08-21:
I think ideally 1 kill should be enough, that is the logic

Ansuel, 2022-08-21:
But now that I think about it... Wonder if a good fix would be call wifi off before... Seems a nice workaround and is not that dirty... (Considering we can make the change for all target)
The system will reboot anyway so no functional change... Just permitting more time for every device to disconnect.

What do you think?

robimarko, 2022-08-21:
Well, that should be true as KILL is supposed to be able to terminate anything, but the one thing it cant do is terminate a part of the kernel obviously.

Yeah, calling wifi off before trying TERM should be a graceful way to sort this out

Ansuel, 2022-08-21:
Will do a pr, lets see if anyone is against this.idea

robimarko, 2022-08-21:
Better first test if that sorts out the issue

Ansuel, 2022-08-21:
https://github.com/Ansuel/openwrt/commit/20f0c87165059cead79b486857ed6d0a1e7830eb ?

robimarko, 2022-08-21:
I dont think it worked as hostapd was still alive and needed SIGTERM

Ansuel, 2022-08-21:
Mhhh wifi down doesn't terminate hostapd. The idea was WiFi removed, all
peer removed quickly (as every device is disconnected). And killing hostapd
should be instant...

robimarko, 2022-08-21:
But it won't work, as wifi uses ubus which AFAIK is not available at the time you are calling it

Ansuel, 2022-08-21:
Mhh we call wifi down before killing ubus if I'm not wrong. This problem is so strange... Are we really sure it's peer delete the problem?

robimarko, 2022-08-21:
Well, I always get the error that it failed flushing the peer or something like that.

Wh1terat, 2022-08-22:
@kirdesde Just me being finicky, but this feels more concise (and less commands)
```bash
enable_affinity_ipq807x(){
        set_affinity(){
            irq=$(awk "/$1/{ print substr(\$1, 1, length(\$1)-1); exit }" /proc/interrupts)
            [ -n "$irq" ] && echo $2 > /proc/irq/$irq/smp_affinity
        }

        #assign 4 rx interrupts to each cores
        set_affinity 'reo2host-destination-ring1' 1
        set_affinity 'reo2host-destination-ring2' 2
        set_affinity 'reo2host-destination-ring3' 4
        set_affinity 'reo2host-destination-ring4' 8

        #assign 3 tcl completions to last 3 CPUs
        set_affinity 'wbm2host-tx-completions-ring1' 2
        set_affinity 'wbm2host-tx-completions-ring2' 4
        set_affinity 'wbm2host-tx-completions-ring3' 8
}
```

Ansuel, 2022-08-22:
@robimarko do we have a way to repro the peer issue in a more sane way?

From what I read in some message killing wpad doesn't trigger the problem...

robimarko, 2022-08-22:
@Ansuel For me it only happens after the AP has been up for some time, so I assume it becomes an issue after you roam to and from a couple of times.

Then I get the `ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 4` and similar

rmandrad, 2022-08-23:
just been testing the latest updates - and noticed that ssdk_sh is not showing as a package anymore. For the qnap & the aquantia phy ssdk_sh although not an absolute requirement is useful to have.

robimarko, 2022-08-23:
Yes, it's not included as I dropped the feed since SSDK and NSS-DP are part of the main tree now.
I must have somehow deleted the release notes as all of that was written.

What do you need it for AQR?

rmandrad, 2022-08-23:
> Yes, it's not included as I dropped the feed since SSDK and NSS-DP are part of the main tree now. I must have somehow deleted the release notes as all of that was written.
> 
> What do you need it for AQR?

I understand this was to enable eth5 on lan @kirdesde ?
ssdk_sh debug phy set 0 0x4004c441 0x8

robimarko, 2022-08-23:
Read through the thread, that is already the default value.
It has all been tested in the forum thread:
https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934/312?u=robimarko

kirdesde, 2022-08-25:
> @Ansuel For me it only happens after the AP has been up for some time, so I assume it becomes an issue after you roam to and from a couple of times.
> 
> Then I get the `ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 4` and similar

The only possibility I see is adding a "killall -9 hostapd 2>/dev/null" to /lib/upgrade/stage2  (line 156 for example)
But that's obviously not target specific.

I haven't find a target specific way to kill processes.

robimarko, 2022-08-25:
Unfortunately, that won't fly for adding the target to OpenWrt.
We either gotta figure out a target specific mitigation or fix ath11k.

kirdesde, 2022-08-25:
> Unfortunately, that won't fly for adding the target to OpenWrt. We either gotta figure out a target specific mitigation or fix ath11k.

Yeah I know, but the problem is, if the ath11k issue occurs, sysupgrade exits before even executing the target specific platform.sh.

So the only way would be a fix for ath11k.....can you point me to the peer patch you mentioned to Ansuel?

robimarko, 2022-08-25:
Think this was the commit:
https://git.codelinaro.org/clo/qsdk/oss/system/feeds/wlan-open/-/commit/f140493181bd2bcaafc11edcf951f68770c063e8

Ansuel, 2022-08-25:
@robimarko I pushed the fixed commit with the real end of the nand. Think you can include that in pr :D

tiagogaspar8, 2022-08-25:
Will it be mandatory from now on to use UART to flash openwrt?

Ansuel, 2022-08-25:
Until someone creates a migration image... Yes :(

tiagogaspar8, 2022-08-25:
Well, I better create a fork of the current pr 😅😅😅😅
I really don't care about the extra space.

robimarko, 2022-08-25:
I mean, you only gotta flash once with UART and that's it.

tiagogaspar8, 2022-08-25:
Yeah, but that might prove too hard for some users who'd don't have that equipment, I mean, if I had the knowledge I'd make a transition image, but unfortunately I do not have it, and for the time being I'm ok with the 32mb limit, I would even suggest the creation of two images, one with the current layout and one with the new expanded layout but I bet that wouldn't fly with upstream 😅😅😅

robimarko, 2022-08-25:
Well, we can't have it both ways here, support for two images ain't gonna fly nor it should, it's a waste of resources.
That is why its getting done before the PR, so there is no need to care about the compatibility mess it would cause later.

edrikk, 2022-08-25:
I don’t think it makes sense to force a UART install.  It limits the number of users (eg I won’t), and in all honesty those who want to add additional packages can do so via image builder so it’s compiled in.

this would just introduce a barrier to people who just want to use the device running OpenWRT.

please let’s not go down this forced UART path.

edrikk, 2022-08-25:
@Ansuel @robimarko as a follow-up to the above, I noticed that nobody replied to this [post](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/7385) 

Given that you have UART connected, can you please validate that forcing the initramfs build type does infact accomplish the same result without UART?  That what this user (LOS) was saying and then seems to have confirmed in response to your question.

> Nope, it was buttery smooth!
On first device was used serial on second enabled serial access just in case and force flashed initramfs image. The result was same on both of them :sunny: so far so happy!
Sysupgraded yesterday both of them as robi rebased his repo and I have applied your commits from increased repo on top of it. It worked like charm!

robimarko, 2022-08-25:
Well, yeah you can boot initramfs image from flash as well if ubinized as that is what U-boot expects.
Bootloader couldn't really care less as long there is a valid uImage there, as there is no difference between just a kernel and DTB in uImage or a kernel that has baked in initramfs(This is what OpenWrt does) and DTB.

The issue is not the default packages, but rather that you are stuck with less than 30MB of usable space out of 256MB.
Given that Xiaomi devices are about as unfriendly to 3rd party SW as it currently is, I dont see this restricting anybody from installing.

FFS, if nobody really wants to put in the time, I will make the one-off conversion image and stick it on the wiki, and be done with it.

edrikk, 2022-08-25:
Honestly, a one time conversion image is the most user friendly way.  So yes please.  And thank you!

Not to belabour the point, but if you’re saying that forcing the initramfs version as part of sysupgrade ends up successfully converting, then why not just have that be it “to convert” and then going forward for upgrades the normal sysupgrade version can be used as usual?

robimarko, 2022-08-25:
You cannot sysupgrade using an initramfs image, for it to work it needs to be a version that has been ubinized and then it can be ubiformated.

That isnt a conversion per se as the rootfs is baked in so the partition table being "wrong" doesnt matter, but yeah that is the most likely route.

milouk, 2022-08-26:
For what it matter I too agree that a one-off conversion would be the best and more user-friendly option

edrikk, 2022-08-26:
> You cannot sysupgrade using an initramfs image, for it to work it needs to be a version that has been ubinized and then it can be ubiformated.

Got it.  So you’re basically saying that the initramfs image that Ansuel built were already ubinized, which is why the above user was able to force sysupgrade the initramfs version.

robimarko, 2022-08-26:
I have no idea what was done to make it sysupgradeable, but even if ubinized it wont work for sysupgrade as there is no metadata.

[19 more replies]

closed 2023-01-16 by robimarko