robimarko/openwrt #56: Can shellsync be add into tree?

gtcar2020, 2022-08-20:
When I attempt to build whith luci-app-syncdial , It notify need shellsync,Can shellsync be add into tree?
Meanwhile I need turboacc also,I have no idea how to add it.

robimarko, 2022-08-20:
Is that part of OpenWrt or out of tree LuCI app?

tm4dev, 2022-08-20:
> When I attempt to build whith luci-app-syncdial , It notify need shellsync,Can shellsync be add into tree?
> 
> Meanwhile I need turboacc also,I have no idea how to add it.
> 
> 

For syncdial, did you try mwan3, luci-app-mwan3?

gtcar2020, 2022-08-20:
yes , I build whith mwan3 and luci-app-mwan3 also.

tm4dev, 2022-08-20:
> Is that part of OpenWrt or out of tree LuCI app?

Out of tree, I think

gtcar2020, 2022-08-20:
It' location just like 'openwrt/tree/xxx/package/network/services/shellsync'

robimarko, 2022-08-20:
Then its not part of OpenWrt, goal of this repo is to get IPQ807x into OpenWrt, not create a fork.

closed 2022-08-20 by robimarko