robimarko/openwrt #62: latest source compiled architecture does not display frequency but display temperature!

zxlhhyccc, 2022-09-05:
latest：
[image]

old code：
[image]

robimarko, 2022-09-05:
Is that LuCI or?

Even if it is, that's a LuCI bug

robimarko, 2022-09-05:
Where is this even shown?

zxlhhyccc, 2022-09-05:
> Where is this even shown?

architecture：
[image]

robimarko, 2022-09-05:
Are you kidding me or what?

Is this LuCI, if so whats the theme, applets etc.
99% this is a generic bug and nothing special to ipq807x

zxlhhyccc, 2022-09-05:
> Are you kidding me or what?
> 
> Is this LuCI, if so whats the theme, applets etc. 99% this is a generic bug and nothing special to ipq807x

Can I add support for IPQ8071-MF269?

robimarko, 2022-09-05:
I am getting annoyed with lack of information here, so one last time before I close this.

HOW TO REPRODUCE THIS "ISSUE"?

zxlhhyccc, 2022-09-05:
> I am getting annoyed with lack of information here, so one last time before I close this.
> 
> HOW TO REPRODUCE THIS "ISSUE"?

Go to Status > Overview > Architecture, shown behind ARMv8

kirdesde, 2022-09-05:
> > I am getting annoyed with lack of information here, so one last time before I close this.
> > HOW TO REPRODUCE THIS "ISSUE"?
> 
> Go to Status > Overview > Architecture, shown behind ARMv8

This must be a customized LuCI version.

LuCI vanilla doesn't show temperature on the status page.

robimarko, 2022-09-05:
Ok, that makes sense.

Closing as its clearly a custom LuCI/package issue.

closed 2022-09-05 by robimarko