robimarko/openwrt #70: Enabling 802.11s breaks all traffic on the radio band

soxrok2212, 2022-09-26:
Hi! I have a fresh build from this morning running on my Q-Hora 301w. I have 4 VLANs and 4 separate SSIDs on both 2.4 and 5GHz. When I enable 802.11s on a new network, all traffic on the associated band breaks completely. I tested with `wpad-mesh-wolfssl` and `wpad-mesh-openssl`, same result on both.

The other radio band is unaffected as long as there is not a mesh network on that same physical radio.

WDS is also broken, perhaps related to https://github.com/robimarko/openwrt/pull/48. With WDS enabled, the whole band will not break but traffic doesn't flow across the wireless link. Other networks on the same radio seem OK.

Thanks for your hard work otherwise!

robimarko, 2022-09-27:
Can you try disabling decap offloading?

soxrok2212, 2022-09-27:
Yep I removed that line, rebuilt and flashed. Still breaking.

soxrok2212, 2022-09-28:
Looks like it's been reported on the forums as well: https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/7993?u=soxrok2212

robimarko, 2022-09-30:
Hm, can you dump FDB using ssdk shell before and after it stops working?

soxrok2212, 2022-09-30:
New terms to me, got a link to explain them? I also got mesh working on 2.4GHz but enabling it on 5 broke the band.

robimarko, 2022-09-30:
Then its probably ath11k FW issue, otherwise it would break on any band

robimarko, 2023-01-16:
Is this still broken?

soxrok2212, 2023-01-16:
Appears to be working fine now!

robimarko, 2023-01-16:
Awesome, thanks