robimarko/openwrt #71: DHCP over B.A.T.M.A.N VLAN bad UDP checksum

soxrok2212, 2022-10-01:
Hi again, facing another issue that wasn't present when I tested between my Redmi AX6000s with identical configs. My setup is:
* Q-Hora 301w as my gateway/router
* Redmi AX6000 as dumb AP (firewall, DHCP disabled)

301w has an 802.11s interface on 2.4GHz with the AX6000 as a peer and a batman interface running across this link. batctl is fine, I can verify connectivity works between them.

Both devices have a `br` bridge with 5 VLANs. bat0 is a member interface and is tagged on all 5 VLANs (DSA). When connecting a client to any one of the VLANs through the dumb AP, I'm unable to get a DHCP address. Assigning a static IP works fine and traffic flows normally. Here's a tcpdump on br.2 (IOT network) on the gateway:

```
00:33:02.167779 xx:xx:xx:xx:xx:xx > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 342: (tos 0x0, ttl 255, id 35097, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from xx:xx:xx:xx:xx:xx, length 300, xid 0x61f4edfd, Flags [none] (0x0000)
	  Client-Ethernet-Address xx:xx:xx:xx:xx:xx
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message (53), length 1: Discover
	    Parameter-Request (55), length 9: 
	      Subnet-Mask (1), Classless-Static-Route (121), Default-Gateway (3), Domain-Name-Server (6)
	      Domain-Name (15), Unknown (108), URL (114), Unknown (119)
	      Unknown (252)
	    MSZ (57), length 2: 1500
	    Client-ID (61), length 7: ether xx:xx:xx:xx:xx:xx
	    Lease-Time (51), length 4: 7776000
00:33:02.168865 yy:yy:yy:yy:yy:yy > xx:xx:xx:xx:xx:xx, ethertype IPv4 (0x0800), length 343: (tos 0xc0, ttl 64, id 42425, offset 0, flags [none], proto UDP (17), length 329)
    10.0.2.1.67 > 10.0.2.16.68: [bad udp cksum 0x1957 -> 0xae74!] BOOTP/DHCP, Reply, length 301, xid 0x61f4edfd, Flags [none] (0x0000)
	  Your-IP 10.0.2.16
	  Server-IP 10.0.2.1
	  Client-Ethernet-Address xx:xx:xx:xx:xx:xx
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message (53), length 1: Offer
	    Server-ID (54), length 4: 10.0.2.1
	    Lease-Time (51), length 4: 43200
	    RN (58), length 4: 21600
	    RB (59), length 4: 37800
	    Subnet-Mask (1), length 4: 255.255.255.0
	    BR (28), length 4: 10.0.2.255
	    Default-Gateway (3), length 4: 10.0.2.1
	    Domain-Name (15), length 3: "lan"
	    Domain-Name-Server (6), length 8: 10.0.0.102,10.0.0.163
00:33:03.180637 xx:xx:xx:xx:xx:xx > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 342: (tos 0x0, ttl 255, id 35098, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from xx:xx:xx:xx:xx:xx, length 300, xid 0x61f4edfd, secs 1, Flags [none] (0x0000)
	  Client-Ethernet-Address xx:xx:xx:xx:xx:xx
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message (53), length 1: Request
	    Parameter-Request (55), length 9: 
	      Subnet-Mask (1), Classless-Static-Route (121), Default-Gateway (3), Domain-Name-Server (6)
	      Domain-Name (15), Unknown (108), URL (114), Unknown (119)
	      Unknown (252)
	    MSZ (57), length 2: 1500
	    Client-ID (61), length 7: ether xx:xx:xx:xx:xx:xx
	    Requested-IP (50), length 4: 10.0.2.16
	    Server-ID (54), length 4: 10.0.2.1
00:33:03.182329 yy:yy:yy:yy:yy:yy > xx:xx:xx:xx:xx:xx, ethertype IPv4 (0x0800), length 343: (tos 0xc0, ttl 64, id 42429, offset 0, flags [none], proto UDP (17), length 329)
    10.0.2.1.67 > 10.0.2.16.68: [bad udp cksum 0x1957 -> 0xab73!] BOOTP/DHCP, Reply, length 301, xid 0x61f4edfd, secs 1, Flags [none] (0x0000)
	  Your-IP 10.0.2.16
	  Server-IP 10.0.2.1
	  Client-Ethernet-Address xx:xx:xx:xx:xx:xx
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message (53), length 1: ACK
	    Server-ID (54), length 4: 10.0.2.1
	    Lease-Time (51), length 4: 43200
	    RN (58), length 4: 21600
	    RB (59), length 4: 37800
	    Subnet-Mask (1), length 4: 255.255.255.0
	    BR (28), length 4: 10.0.2.255
	    Default-Gateway (3), length 4: 10.0.2.1
	    Domain-Name (15), length 3: "lan"
	    Domain-Name-Server (6), length 8: 10.0.0.102,10.0.0.163
```

The ACK never makes is to the br.2 interface on the dumb AP, presumably because of the bad UDP checksum on that message. I found this that's related, but is a fix for old iptables: https://forum.openwrt.org/t/solved-dhcp-on-vlan-interfaces/61115/5

Also for reference: https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934/446?u=soxrok2212

robimarko, 2022-10-04:
I am gonna be honest, this is beyond me but I still find it weird that UDP checksum is broken as its all calculated in SW and not offloaded and thus should be always correct

soxrok2212, 2022-10-04:
FWIW this only happens on wireless. Running Batman over a wire doesn’t have this issue.

robimarko, 2022-10-04:
Hm, have you tried disabling decap offloading?

soxrok2212, 2022-10-10:
Disabling decapsulation did not fix the issue.

robimarko, 2023-01-16:
Is this still broken?

soxrok2212, 2023-01-16:
Appears to be working now

closed 2023-01-16 by soxrok2212

robimarko, 2023-01-16:
Great, thanks