robimarko/openwrt #72: qca-ssdk 2021-04-28 build failed with gcc 12: adpt_hppe_portvlan.c:1719:9: error: 'entry_idx' may be used uninitialized

lovehackintosh, 2022-10-03:
Hello.I am not good at speaking English.

qca-ssdk 2021-04-28 build failed with gcc 12: adpt_hppe_portvlan.c:1719:9: error: 'entry_idx' may be used uninitialized[-Werror=maybe-uninitialized]

My config is ipq8071/ax3600,kernel5.15 and gcc12

<details>
<summary>log</summary>
<pre><code>  
&nbsp;
make[6]: Entering directory '/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/src/adpt/hppe'
/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc  -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/common -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/api -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/fal -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/ref -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/adpt -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/hsl -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/hsl/phy -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/sal/os -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/sal/os/linux -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/sal/sd -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/sal/sd/linux/hydra_howl -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/sal/sd/linux/uk_interface -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/init -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/hsl/hppe -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/adpt/hppe -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/include/shell_lib -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71 -I/home/clover/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/../lib/gcc/arm-openwrt-linux-musl/12.2.0/include/ -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/source -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/source/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/source/arch/arm64/mach-msm/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/arch/arm64/mach-msm/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/source/arch/arm64/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/arch/arm64/include -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/source/arch/arm64/include/asm -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/arch/arm64/include/generated -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/include/generated/uapi -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/include/uapi -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/arch/arm64/include/uapi -I/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/source/arch/arm64/include/asm/mach -include /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.71/include/linux/kconfig.h -DIN_ACL -DIN_FDB -DIN_IGMP -DIN_LEAKY -DIN_LED -DIN_MIB -DIN_MIRROR -DIN_MISC -DIN_PORTCONTROL -DIN_PORTVLAN -DIN_QOS -DIN_RATE -DIN_STP -DIN_VLAN -DIN_COSMAP -DIN_IP -DIN_NAT -DIN_FLOW -DIN_TRUNK -DIN_SEC -DIN_QM -DIN_INTERFACECONTROL -DIN_CTRLPKT -DIN_SERVCODE -DIN_RSS_HASH -DIN_MALIBU_PHY -DIN_AQUANTIA_PHY -DIN_QCA803X_PHY -DIN_QCA808X_PHY -DIN_SFP_PHY -DIN_PHY_I2C_MODE -DIN_VSI -DIN_PPPOE -DIN_BM -DIN_SHAPER -DIN_POLICER -DIN_UNIPHY -DUK_IF -DUK_NL_PROT= -DUK_MINOR_DEV=254  -Wall -DVERSION=\"2.0.0.2\" -DBUILD_DATE=\"2022-10-02-14:45:42\" -DOS=\"linux\" -D"KBUILD_STR(s)=#s" -D"KBUILD_MODNAME=KBUILD_STR(qca-ssdk)" -DHPPE -DKVER34 -DKVER32 -DLNX26_22 -D__KERNEL__ -DKERNEL_MODULE -DMODULE -Os -pipe -march=armv8-a -mcpu=cortex-a53+crypto -fno-caller-saves -fno-strict-aliasing -Werror -fno-common -Wno-format-security -Wno-pointer-sign -Wno-unused-but-set-variable -Wno-error=unused-result -mcmodel=large -fno-stack-protector -c adpt_hppe_portvlan.c -o /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/build/linux/KSLIB/adpt_hppe_portvlan.o
adpt_hppe_portvlan.c: In function 'adpt_hppe_port_vlan_trans_adv_add':
adpt_hppe_portvlan.c:1719:9: error: 'entry_idx' may be used uninitialized [-Werror=maybe-uninitialized]
 1719 |         _insert_vlan_trans_adv_rule_action(dev_id, entry_idx, direction, rule, action);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
adpt_hppe_portvlan.c:1644:20: note: 'entry_idx' was declared here
 1644 |         a_uint32_t entry_idx, entry_sign, rule_valid;
      |                    ^~~~~~~~~
cc1: all warnings being treated as errors
make[6]: *** [/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/make/target.mk:13: adpt_hppe_portvlan.o] Error 1
make[6]: Leaving directory '/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/src/adpt/hppe'
make[5]: *** [/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/make/target.mk:7: obj] Error 1
make[5]: Leaving directory '/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/src/adpt'
make[4]: *** [Makefile:27: kslib_o] Error 1
make[4]: Leaving directory '/home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3'
make[3]: *** [Makefile:127: /home/clover/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-nohnat/qca-ssdk-2021-04-28-c9bc3bc3/.built] Error 2
make[3]: Leaving directory '/home/clover/lede/package/qca/qca-ssdk'
time: package/qca/qca-ssdk/nohnat/compile#0.71#0.05#0.76
    ERROR: package/qca/qca-ssdk failed to build (build variant: nohnat).
make[2]: *** [package/Makefile:116: package/qca/qca-ssdk/compile] Error 1
make[2]: Leaving directory '/home/clover/lede'
make[1]: *** [package/Makefile:110: /home/clover/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/clover/lede'
make: *** [/home/clover/lede/include/toplevel.mk:230: world] Error 2
</code></pre>
</details>

How to fix it?

robimarko, 2022-10-03:
What tree are you buliding?

I have a feeling its not from this repo at all as `qca-ssdk 2021-04-28` has been updated long time ago

XiaoliChan, 2022-10-03:
U are using this [Chinese maintainer version](https://github.com/coolsnowwolf/lede), not robi's branch.

lovehackintosh, 2022-10-03:
[coolsnowwolf/lede/package/qca/qca-ssdk](https://github.com/coolsnowwolf/lede/tree/master/package/qca/qca-ssdk) 
It looks from your branch [AX3600-5.10-NSS-11.4](https://github.com/robimarko/openwrt/tree/AX3600-5.10-NSS-11.4) .

robimarko, 2022-10-03:
Then, feel free to open the ticket on that repo, its got nothing to do with this tree.

closed 2022-10-03 by lovehackintosh