robimarko/openwrt #74: hcxdumptool crashes ath11k

soxrok2212, 2022-10-11:
Hi again :) Hope you’re happy to hear from me again. I’m doing some security work and it appears that hcxdumptool crashes ath11k when it hits my network. There’s a variety of things it does (probe request, deauth, ask for pmkid, etc). As soon as I fire it off against my network I see this in dmesg (apologize for crap screenshot, on mobile atm). 

[image]

robimarko, 2022-10-11:
This is something you are gonna have to open a bugzilla ticket about, I dont see what we can do about it.

robimarko, 2023-01-16:
This needs to be opened upstream

closed 2023-01-16 by robimarko