robimarko/openwrt #76: ipq8072: label red led correctly as red for Dynalink DL-WRX36

DonKult, 2022-10-21:
Most of the dts file for the Dynalink DL-WRX36 refers to the led as being red and the specification in the device commit says
as much, but it goes on to label it "yellow" which is what ends up displayed in `/sys`, LuCI and co.

I am happy to convert this to a PR if you prefer, but I suppose this is squashed more easily by hand.

```diff
--- a/target/linux/ipq807x/files/arch/arm64/boot/dts/qcom/ipq8072-dl-wrx36.dts
+++ b/target/linux/ipq807x/files/arch/arm64/boot/dts/qcom/ipq8072-dl-wrx36.dts
@@ -60,9 +60,9 @@
                };

                led_system_red: system-red {
-                       label = "yellow:system";
+                       label = "red:system";
                        gpios = <&tlmm 25 GPIO_ACTIVE_HIGH>;
-                       color = <LED_COLOR_ID_YELLOW>;
+                       color = <LED_COLOR_ID_RED>;
                };
        };
 };
```

Still working on setting up the device for me, but so far I am pleased by the device and your work on supporting it and the whole target-family in OpenWrt. Thanks! :+1:

robimarko, 2022-10-21:
Thanks for catching this, yeah its easier to manually add and squash, will do it tomorrow.

Thanks again, regards.
Robert

robimarko, 2022-10-25:
Thanks, added this and pushed.

closed 2022-10-25 by robimarko