robimarko/openwrt #83: Which branch is the lastest stable for rb450gx4?

swordow, 2022-11-06:
I found that there are some branches for rbg450gx4 and which one is the lastest stable for it ? And if it contains the sd card supports?

robimarko, 2022-11-06:
None really, I dont have the board anymore.

closed 2022-11-06 by robimarko

swordow, 2022-11-06:
> None really, I dont have the board anymore.

Oh...bad news..ahh.
And which branch is better that i could continue developing on for rb450gx4 ?
Thanks.

robimarko, 2022-11-06:
None of mine, john-tho has much newer stuff for RB450Gx4 and support for the new NAND revision
https://github.com/john-tho/openwrt

swordow, 2022-11-07:
> None of mine, john-tho has much newer stuff for RB450Gx4 and support for the new NAND revision https://github.com/john-tho/openwrt

Thanks. I will try it~~