robimarko/openwrt #84: AX9000 PCI WIFI not work, cannot connect.

dick902, 2022-11-11:
PCI 5G wifi issues with latest build (2022-11-05-2125)
The PCI 5G tested with 36 - 64 channel in both 80Mhz and 160Mhz, country was in US, AHB 5G ON or OFF. With WPA3 or no encryption key. AHB 5G and PCI 5G in different SSID.
SSID can be shown up on client devices but none of them can connect with the PCI 5G wifi.
I had tested with 3 different client devices, all resulted the same, none of them can connect with PCI 5G.
The AHB 5G is ok on all test (Channel 149, 80Mhz Country in US).
Seems something wrong with the PCI 5G!
I also tried to switch back to the OEM AX9000 xiaomi factory firmware and confirm the PCI 5G hardware is working just fine with the oem factory firmware.
The PCI 5G was configured as "phy3-ap0", but every time I try to connect with the PCI 5G the logs show "phy1-ap0" activities, and this was supposed to be AHB 5G interface.

robimarko, 2022-11-12:
I am aware of this, its an issue with the stock BDF where you cannot connect, a "workaround" would be to use a generic upstream BDF

dick902, 2022-11-12:
Thx for your advice!
Seems there is no easy fix at this monent!

robimarko, 2022-11-12:
Yeah, its a mistery why the BDF is not working

tiagofreire-pt, 2022-12-17:
A workaround is available and documented on the AX9000 wiki: https://openwrt.org/inbox/toh/xiaomi/ax9000#potential_issueslimitations

closed 2022-12-17 by dick902

Serpentian, 2024-09-14:
The link in the docs to https://textbin.net/ekk9nz46lw is not available anymore. Does anyone have the command to generate a new BDF board-2.bin?