robimarko/openwrt #86: MHI_CB_EE_SBL_MODE not defined

kristrev, 2022-11-24:
Hello,

I am trying to compile an image for my device, based on the `ipq807x-5.15-pr` branch. During compilation, building `mac80211` fails with the error that `MHI_CB_EE_SBL_MODE` is undeclared. I checked upstream and `ath-next`, but the only place I can find any reference to `MHI_CB_EE_SBL_MODE` is in the `104-wifi-ath11k-use-unique-QRTR-instance-ID.patch` patch. Is there a patch missing and which introduces this constant, or am I missing something else?

Removing the patch produces an image that works well, but I thought I should ask :)

Thanks.

robimarko, 2022-11-24:
What is the commit hash you are building off?

Cause its building fine for me as well as Github actions

kristrev, 2022-11-24:
My build is based off 49ccb29. However, I see that I forgot to include a potentially important detail in my original message. My device is an x86-board with an ath11k card, so select the ath11k-pci package (and not ahb).

robimarko, 2022-11-24:
Oh, I get the issue now since you mention its x86, the issue is that MHI patch is only present in ipq807x:
https://github.com/robimarko/openwrt/blob/ipq807x-5.15-pr/target/linux/ipq807x/patches-5.15/0138-bus-mhi-core-add-SBL-state-callback.patch

So, you can try adding it to x86 as well, that should sort it out

kristrev, 2022-11-25:
Thanks, that worked. I must have forgotten to grep through the target-folder.

closed 2022-11-25 by kristrev