robimarko/openwrt #88: Compile Helloworld for IPQ807x

FillHK, 2022-11-29:
Hi, 

Is it possible to compile just one package for IPQ807x? ( this one for example: https://github.com/nicanor-romero/openwrt_helloworld)
So far I have always:
```
make[1]: Entering directory '/home/ow/openwrt_ipq/openwrt'
make[1]: *** No rule to make target 'packages/helloworld/compile'.  Stop.
make[1]: Leaving directory '/home/ow/openwrt_ipq/openwrt'
make: *** [/home/ow/openwrt_ipq/openwrt/include/toplevel.mk:230: packages/helloworld/compile] Error 2
```

joba-1, 2022-11-29:
yes,

but hard to tell how you got there and what’s missing.

These are my notes for compiling my [ethic](https://github.com/joba-1/Ethic) repo for openwrt. Maybe it helps

```
cd ~/git/jobawrt
make toolchain/install
mkdir -p ~/openwrt-feed/net/ethic
# Content of Makefile in attachments
vi ~/openwrt-feed/net/ethic/Makefile
echo "src-link joba $HOME/openwrt-feed" >>feeds.conf
./scripts/feeds update joba
./scripts/feeds install -a -p joba
make menuconfig
make package/ethic/compile
ls -la bin/packages/*/joba/
scp bin/packages/*/joba/ethic_*.ipk ax1:/tmp
ssh ax1 opkg install /tmp/ethic_*.ipk
ssh ax1 ethic -h
```

more details in [the openwrt wiki](https://openwrt.org/docs/guide-developer/toolchain/single.package)

FillHK, 2022-11-29:
It will help me.
One last question: 
* Can I use Ubuntu? What is the version suggested?

joba-1, 2022-11-29:
I don't use it, but why not. The build system is pretty much self contained and worked fine out of the box on my opensuse distro.

FillHK, 2022-11-30:
I was able to compile the package based on your link.
But unable to install it on targeted platform: 
root@OpenWrt:/# cat /etc/openwrt_release
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Chaos Calmer'
DISTRIB_REVISION='bb23379fd+r49254'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='ipq/ipq807x_64'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05.1'
DISTRIB_TAINTS='no-all busybox override'
```
```
 opkg install ethic_1.0-2_aarch64_cortex-a53.ipk
Unknown package 'ethic'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for ethic found, but inc                                    ompatible with the architectures configured
```

robimarko, 2022-11-30:
No, you cannot install it on the stock QSDK

joba-1, 2022-12-02:
looks like you „forgot“ changes in menuconfig:

Target System (Qualcomm IPQ807x)
Subtarget (Generic)
Target Profile (Xiaomi AX3600) (or whatever your device is…)

closed 2022-12-13 by robimarko