robimarko/openwrt #91: Installing Wireguard-tools-> missing kmod-wireguard

naltabra, 2022-12-31:
I tried to report this problem at wrt github but was promply rejected.

Just try yo install wireguard-tools and install tells me that kmod.wireguard is missing.

same problem in Luci and shell.

robimarko, 2022-12-31:
Update to the latest release and then it will work, there is now a feed for the latest release with nonshared packages and kmods

naltabra, 2023-01-01:
I can’t find any newer release than 2350 and thats the release am using.

robimarko, 2023-01-01:
And what is the error?

Have you updated the feeds first?

naltabra, 2023-01-01:
Yes, and now I see a problem. Can you fix it?

Downloading https://feed.robimarko.eu/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/ipq807x_core
Downloading https://feed.robimarko.eu/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.

robimarko, 2023-01-01:
Again, update to the latest release and make sure that the version is: `r0-63ba272`

rodrigoscoelho, 2023-01-01:
I'm using this version, but, no internet access when using as access point
with the router cable connected in the Lan port (not wan).
the IP address is fixed in 192.168.1.17 and the gateway is set to
192.168.1.1

I can access the router, but can't ping from inside it, the internet.

Any advice.

Thanks for your work

robimarko, 2023-01-01:
Really wrong place for that question, better ask on the forum as there is way more users.

naltabra, 2023-01-01:
Not working!

I’m running latest version from about 18:00 OpenWrt SNAPSHOT r0-9594846 / LuCI Master git-22.351.50506-a27d2db.

SIG-file is still broke as mentioned above, so lucy skip your feed and installation won’t work.

You can se for your self. Try to install packet wireguard-tools and dependency will fail because kmod-wireguard lives in customfeeds.conf which is broke (or custom feeds conf function is broke)

The strange thing is that I have another ax3600 installed and configured prior 28/12-version and that one is working as i should, so what happened 28/12?

robimarko, 2023-01-01:
Sorry, but its working:
```
root@OpenWrt:/# opkg update
Downloading https://feed.robimarko.eu/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/ipq807x_core
Downloading https://feed.robimarko.eu/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.gz, wget returned 8.
root@OpenWrt:/# opkg install wireguard-tools
Installing wireguard-tools (1.0.20210914-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/wireguard-tools_1.0.20210914-1_aarch64_cortex-a53.ipk
Installing kmod-crypto-lib-chacha20 (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-crypto-lib-chacha20_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-crypto-lib-poly1305 (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-crypto-lib-poly1305_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-crypto-lib-chacha20poly1305 (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-crypto-lib-chacha20poly1305_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-crypto-kpp (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-crypto-kpp_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-crypto-lib-curve25519 (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-crypto-lib-curve25519_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-udptunnel4 (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-udptunnel4_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-udptunnel6 (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-udptunnel6_5.15.85-1_aarch64_cortex-a53.ipk
Installing kmod-wireguard (5.15.85-1) to root...
Downloading https://feed.robimarko.eu/packages/kmod-wireguard_5.15.85-1_aarch64_cortex-a53.ipk
Configuring kmod-crypto-lib-chacha20.
Configuring kmod-crypto-lib-poly1305.
Configuring kmod-crypto-lib-chacha20poly1305.
Configuring kmod-udptunnel4.
[   59.831879] kmodloader: loading kernel modules from /etc/modules.d/*
[   59.841395] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   59.841433] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   59.848595] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring kmod-udptunnel6.
[   59.880385] kmodloader: loading kernel modules from /etc/modules.d/*
[   59.882964] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring kmod-crypto-kpp.
[   59.909866] kmodloader: loading kernel modules from /etc/modules.d/*
[   59.912463] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring kmod-crypto-lib-curve25519.
Configuring kmod-wireguard.
[   59.957173] kmodloader: loading kernel modules from /etc/modules.d/*
[   59.959760] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring wireguard-tools.
```

You are not running `r0-63ba272` which is the latest release!

naltabra, 2023-01-02:
well it's not working for mee, and I might know what the problem is.
This sysupgrade function is not working as it should.

I'm on the wrong version after upgrading (without error messages). How is that possible? 

I have tried doing Factory reset. I have tried upgrade to squashfs-factory and even force an upgrade but I'm still having BUILD_ID="r0-9594846". And If I look into /etc, my time machine is stuck at "Dec 21 21:05" after upgrade.

How do I go back to square one?

robimarko, 2023-01-02:
Just try stopping wpad before upgrading, and do not try to use the factory image, only sysupgrade image.
Also, networking interface names have changed and so did the compact version, so you can only upgrade without preserving config

naltabra, 2023-01-02:
Nope, same same still at r0-9594846 (21/12), so I tried to go the smallest step upwards, 2022-12-23-2049, - didn't work.

Then I noticed the output from sysupgrade. I'm using Putty. Look below. Can you see something strange?
sysupgrade is a shell script running other parts.. First I thought it was putty throwing "closing messages", but now I think at least one of messages comes from the shell script, resulting in "Command failed". If you look in the snippet there are two cadidates faileing, json_xxx or ubus?

So what do you think?
---

root@OpenWrt:/tmp# sysupgrade -n "openwrt-ipq807x-generic-xiaomi_ax3600-squashfs-sysupgrade (6).bin"
Mon Jan  2 12:22:38 UTC 2023 upgrade: Commencing upgrade. **_Closing all shell sessions._**
Command failed: Connection failed

---
last part of /sbin/sysupgrade script

install_bin /sbin/upgraded
v "Commencing upgrade. _****Closing all shell sessions**.**"_

COMMAND='/lib/upgrade/do_stage2'

if [ -n "$FAILSAFE" ]; then
        printf '%s\x00%s\x00%s' "$RAM_ROOT" "$IMAGE" "$COMMAND" >/tmp/sysupgrade
        lock -u /tmp/.failsafe
else
        json_init
        json_add_string prefix "$RAM_ROOT"
        json_add_string path "$IMAGE"
        [ $FORCE -eq 1 ] && json_add_boolean force 1
        [ $SAVE_CONFIG -eq 1 ] && json_add_string backup "$CONF_TAR"
        json_add_string command "$COMMAND"
        json_add_object options
        json_add_int save_partitions "$SAVE_PARTITIONS"
        json_close_object

        ubus call system sysupgrade "$(json_dump)"
fi
root@OpenWrt:~#

robimarko, 2023-01-02:
Sysupgrade will stop `dropbear` so SSH wont work, that is normal.

Do you have UART maybe, cause there is plenty of users that were able to upgrade just fine?

robimarko, 2023-01-04:
Without UART there is no way to see where it gets stuck as all of the stuff happens after SSH is dead.

naltabra, 2023-01-05:
Now I'm back on track after spending some good time with tftp-recovery. Now I'm running the latest version on this one.

But as I told you, I have two ax3600 that will become a wire pair. This other one is running another version from start: 
OpenWrt SNAPSHOT **r0-4203028** / LuCI Master git-22.361.69865-deed682

Do you think it will upgrade? NO!! 
Exactly the same problem. Upgrading seems to go well but when checking still on the same version. 
This is a basic function and should just work! You have a BUG.

robimarko, 2023-01-05:
Again, without a UART log, I have no idea where it fails as its working for me and others.
So, how about instead of demanding that it must work, you provide some debug info?

l33tlinuxh4x0r, 2023-01-09:
I have noticed that sysupgrade doesn't work about 50% of the time. I use the web interface to try and flash the router, it appears to flash, the leds turn red etc but upon reboot it does just that reboot. If I flash it again about a minute later (or less) it sticks. So pretty much I just have to flash it twice. I noticed this on windows I will see if the issue persists on linux. I know when I get a successful flash because the version number changes and my installed software packages are removed... like upnp, printer stuff, ddns etc.

robimarko, 2023-01-16:
Target has been merged into OpenWrt, there should be prebuilt images soon.