robimarko/openwrt #94: LEDS for ports LAN 1-4 are backwards with build 2023-01-03-1333

l33tlinuxh4x0r, 2023-01-04:
I noticed that 10g-1 is now the wan port by default👍
10G port LEDs are working correctly.
LAN ports 1-4 on the back of the device correspond to the opposite LED.
I'm using the qnap 301w with build  2023-01-03-1333
```
Port: 1-2-3-4
LED: 4-3-2-1
```

l33tlinuxh4x0r, 2023-01-04:
Also the wifi LED isn't enabled for any device by default is this on purpose?

robimarko, 2023-01-05:
Thanks for catching this, Il get it sorted out probably today.

There was not WLAN trigger before, I need to check whether it exists now.

robimarko, 2023-01-06:
I checked why WLAN LED is not set, and well there is only one LED for 2 radios so its up to users to configure it to light on one of the radios.

I have fixed the LAN LED-s will be pushed today.

l33tlinuxh4x0r, 2023-01-06:
As far as the WLAN LED goes would it be possible to make vlan or bridge with both radios going through it and then set the LED to that?

robimarko, 2023-01-06:
Not really, anyway LED fix has been pushed.

closed 2023-01-06 by robimarko