robimarko/openwrt #99: Currently how to write SFP ports in dts file of ipq60xx and ipq807x?

qlugcp, 2023-02-02:
### Describe the bug

https://github.com/robimarko/openwrt/blob/ipq807x-5.15-pr-nss-drv/package/kernel/qca-nss-dp/patches/0013-nss_dp_main-Use-a-phy-handle-property-to-connect-to-.patch

Currently this patch have removed the QCA specific property and assume that all ports will use mdio bus to access port info, BUT **SFP** ports do not use mdio bus , instead, it use I2c to get port info from that, so how to handle this in dts file?

### OpenWrt version

master

### OpenWrt target/subtarget

ipq807x&ipq60xx

### Device

ipq807x

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-02-02:
Currently, there isn't really a way, hence the driver needs further fixup-s to respect the SFP property and attach it properly.

Which device do you have with SFP?

qlugcp, 2023-02-02:
Tplink [TL-ER2260T](https://www.tp-link.com.cn/product_2295.html?source=list) which is IPQ8070 with 4*1000Mbps rj45 + 2*10GE SFP+ port and **no wireless** and I have add basic support [here](https://github.com/coolsnowwolf/lede/pull/10777)

qlugcp, 2023-02-02:
> Currently, there isn't really a way, hence the driver needs further fixup-s to respect the SFP property and attach it properly.
> 
> Which device do you have with SFP?

Tplink [TL-ER2260T](https://www.tp-link.com.cn/product_2295.html?source=list) which is IPQ8070 with 4*1000Mbps rj45 + 2*10GE SFP+ port and **no wireless** and I have add basic support [here](https://github.com/coolsnowwolf/lede/pull/10777)

robimarko, 2023-02-07:
Oh, I have been watching those TP-Link router-only models but they are all CN-only.
I really dont have a way to try SFP support at all as none of my HW has one

robimarko, 2023-10-15:
Look at my WIP RT-AX89X work to see how to wire up a SFP port.
https://github.com/robimarko/openwrt/tree/rt-ax89x

closed 2024-02-19 by robimarko