robimarko/openwrt #100: 5GHz temp instantly increases after performing a SCAN

AlexeiGHub, 2023-03-12:
### Describe the bug

Hello!
Perhaps with this problem it was worth going to the openwrt forum, but my hopes for a solution to this bug by you are higher.
5GHz temp of ath11k_hwmon / temp1 instantly increases after performing a SCAN from 51°C to C71 °C  

### OpenWrt version

r22257-166ab6f90e

### OpenWrt target/subtarget

ipq807x/generic

### Device

Redmi AX6

### Image kind

Official downloaded image

### Steps to reproduce

Just perfome Scan AP in Luci on 5GHz(radio0)

### Actual behaviour

Temperature of ath11k_hwmon / temp1	increase from 51°C to C71 °C instantly 

### Expected behaviour

Temperature must still the same 51°C

### Additional info

Log in "Emergency" mode nothing print at this time.
For monitor temp I use this https://github.com/gSpotx2f/luci-app-temp-status
Problem first notised on official openwrt snapshot at 05.03.2023 beacose  I install uci-app-temp-status
Promblem available on clear install openwrt snapshot with simple one AP

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-03-12:
Why do you think that a temperature increase once the radio is actually being used is worrying?

AlexeiGHub, 2023-03-12:
It used before but after scan  rise temperature and still high.
Or if it used as STA and happend disconnect and then connect again it rise.
Clear check 1 AP on radio temperature 51deg. but after scan 71? 
I think I think this is abnormal and indicates a problem.

But after reboot AP work at temperature 51deg untill I perfome SCAN

robimarko, 2023-03-12:
I still dont understand what is the issue and how is it an OpenWrt or even a kernel issue.
71 degrees is not high at all

AlexeiGHub, 2023-03-12:
Surely 71 degrees is acceptable.
I just found this oddity how to reproduce it and how to observe the consequences.

And I assume that after that the memory leak starts.

Perhaps the mechanism itself and the manifestation will interest you

AlexeiGHub, 2023-03-20:
I have ax6s router. It have similar name but different hw.
In LUCI order radio differ from ax6. First 2.4G and second 5G.
I see temp of both radio phy before scan and 2.4 was much less then 5G. I was connected to 2.4G. 
At first I perfome SCAN on 5G(wl1) and no change temperature effect but after SCAN on wl0(2.4G) phy0 temperature permanently rise on +10deg. 
Amazing effect, do you agree?

AlexeiGHub, 2023-03-20:
Did you reproduce this feature on your hardware?

robimarko, 2023-03-20:
No, I did not attempt to reproduce it as temperatures are reasonable

AlexeiGHub, 2023-04-09:
With 2.9.0.1 FW  no change :( Temp rise as before.

robimarko, 2023-10-15:
I dont consider this an issue.

closed 2023-10-15 by robimarko