robimarko/openwrt #101: I can't install openvpn-openssl package

diegoweb, 2023-03-16:
### Describe the bug

Hi!!
I've just flashed my AX6 with:
`https://github.com/robimarko/openwrt/releases/download/ipq807x-2023-01-14-1709/openwrt-ipq807x-generic-redmi_ax6-squashfs-sysupgrade.bin`

And I'm unable to install openvpn-openssl package. It looks like the package requires a newer kernel version, and there's no newer version of openwrt in your repository to download.

### OpenWrt version

r0-3a2144f

### OpenWrt target/subtarget

ipq807x/generic

### Device

Redmi AX6

### Image kind

Official downloaded image

### Steps to reproduce

`opkg update`
`opkg install openvpn-openssl`

### Actual behaviour

```
root@Escritorio:~# opkg install openvpn-openssl
Installing openvpn-openssl (2.5.8-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/openvpn-openssl_2.5.8-1_aarch64_cortex-a53.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.98-1-caaef7c842ee3a8bac7a225270cddab0) for kmod-tun
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-tun found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for openvpn-openssl:
 *      kernel (= 5.15.98-1-caaef7c842ee3a8bac7a225270cddab0)
 * opkg_install_cmd: Cannot install package openvpn-openssl.
```

### Expected behaviour

Package getting installed without any errors.

### Additional info

I've tried to install kmod-tun to see if I was going to get a different error, but no luck:

```
root@Escritorio:~# opkg install kmod-tun
Unknown package 'kmod-tun'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.98-1-caaef7c842ee3a8bac7a225270cddab0) for kmod-tun
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-tun found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-tun.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-03-16:
All of those releases are completely unsupported as everything was upstreamed, just use upstream OpenWrt snapshots

diegoweb, 2023-03-16:
> All of those releases are completely unsupported as everything was upstreamed, just use upstream OpenWrt snapshots

Ohhhhh! I thought that since the gist available in the AX6 wiki was updated in Jan, 2023, it was the latest release.

So I found the correct image you mentioned:
https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/openwrt-ipq807x-generic-redmi_ax6-squashfs-sysupgrade.bin

But when I try to do a sysupgrade, I get this error message:
[image]

Should I force upgrade?

robimarko, 2023-03-16:
If you are on compat 2.0 then forcing should work

diegoweb, 2023-03-16:
> If you are on compat 2.0 then forcing should work

Nice! It worked.
But it looks like those snapshots doesn't have LuCi, right?

I can't browse to 192.168.1.1 even after reboot. But I can SSH fine to the router.

robimarko, 2023-03-16:
Yes, by default snapshots dont include luci, but you can install it via SSH by using opkg

diegoweb, 2023-03-16:
Nice, thank you very much robimarko!

closed 2023-03-16 by diegoweb